Simsso/ShiftRegister74HC595 #2: Shift register pulsing on and off whenever delay is used after "set()"

stevepeart, 2017-10-31:
Hello,

I am having an issue when using this library with delays, including the example code.

I have 3 LEDs connected to the shift register outputs and whenever I call:

sr.set(0, HIGH);
sr.set(1, HIGH);
sr.set(2, HIGH);

In my main loop, the 3 lights all come on like expected.

However, when I change the code to this:

sr.set(0, HIGH);
delay(1000);
sr.set(1, HIGH);
delay(1000);
sr.set(2, HIGH);
delay(1000);

I can see the lights come on with one second delays, but the lights just pulse on and off, rather than have each one light up with a one second delay between like expected. Is this a bug?

Simsso, 2017-11-02:
Hi Stevepeart, thanks for reporting the issue. 
Before I wire things up this weekend in order to try to reproduce the error, I wanted to ask whether you have double-checked the wiring. Does the issue persist?

```
int serialDataPin = 0; // DS
int clockPin = 1; // SHCP
int latchPin = 2; // STCP
ShiftRegister74HC595 sr (numberOfShiftRegisters, serialDataPin, clockPin, latchPin);
```

[image]

Simsso, 2017-11-04:
Hi @stevepeart 
unfortunately I was not able to reproduce the bug with the following code:
```C++
#include <ShiftRegister74HC595.h>

// create shift register object (number of shift registers, data pin, clock pin, latch pin)
ShiftRegister74HC595 sr (1, 2, 3, 4); 
 
void setup() { }

void loop() {
  sr.setAllLow(); // set all pins LOW
  delay(500); 
  sr.set(0, HIGH);
  delay(1000);
  sr.set(1, HIGH);
  delay(1000);
  sr.set(2, HIGH);
  delay(1000); 
}
```

It works as expected. Make sure your wiring is correct and don't transmit data over `Serial` when using the pins 0 and 1 for your shift register too.

Regards,
Timo

closed 2017-11-04 by Simsso

stevepeart, 2017-11-07:
You are right, I had pin 11 and 12 reversed on the shift register. Now I am getting results I expect, thanks!