Simsso/ShiftRegister74HC595 #7: The library currently isn't working for me due to latch conditions

hedrickbt, 2018-03-19:
My 74HC595 chip expects the latching to be LOW instead of high.  I have provided a pull request with the changes I needed.  

https://github.com/Simsso/Shift-Register-74HC595-Arduino-Library/pull/6

Simsso, 2018-03-20:
Hey @hedrickbt,

thanks a lot for your contribution. It's greatly appreciated.

I do not have access to any hardware right now but I'll test your changes as soon as possible (approx. two weeks from now). Then I can merge it.

hedrickbt, 2018-04-01:
Added some questions to https://github.com/Simsso/ShiftRegister74HC595/pull/6

Simsso, 2018-04-01:
_Let's move this conversion here. Makes more sense to discuss the issue rather than the PR._

I have looked into the suggested changes is detail now and there are a few things not clear to me. Could you please clarify:

- You set the latch pin to `LOW` before shifting out new data (https://github.com/Simsso/ShiftRegister-74HC595-Arduino/pull/6/commits/032c9b85ed5734805a80aa397359440902a0401a#diff-9517b5fd53eb127cde55dd0bd709ca34R36). There is (to the best of my knowledge) no reason to do so.
- No matter whether your shift register is sensitive to changes from `LOW` to `HIGH` or vice-versa, both signal edges are being output by the original version as well. https://github.com/Simsso/ShiftRegister-74HC595-Arduino/blob/master/ShiftRegister74HC595.cpp#L42

closed 2018-04-12 by Simsso