Simsso/ShiftRegister74HC595 #16: getAll() for multiple shift registers

lnorton89, 2019-04-03:
Maybe I'm not reading the documentation correctly but how would one get the status of all shift register outputs to print to the Serial console?

Simsso, 2019-04-04:
Hi @lnorton89:
One way of achieving this is the following:

1. Retrieve the values with `uint8_t * vals = sr.getAll()`.
2. Loop over the number of shift registers, counter variable `i`
3. For each shift register call `Serial.println(vals[i], BIN)`

Please close the issue if this answers your question.

lnorton89, 2019-04-04:
I have two shift registers the first one being for outputs the second one for inputs. Would I loop from 9-16 to retrieve the inputs?

Simsso, 2019-04-04:
The code looks good, however, you cannot use shift register pins like input pins. That's simply not possible with the 74HC595.

lnorton89, 2019-04-04:
I was under the assumption it was possible. It would be nice to use a library like yours to read inputs instead of the SPI library. 

https://www.youtube.com/watch?v=nXl4fb_LbcI

Simsso, 2019-04-04:
Thank you for pointing this out. I'm a bit surprised, also because the [data sheet](https://shiftregister.simsso.de/data-sheet.pdf) clearly states that the pins are outputs.

closed 2019-04-04 by Simsso