Simsso/ShiftRegister74HC595 #21: Sending data to multiple registers

yuvalkedar, 2020-08-12:
Hi,
I couldn't understand how do I send different array for each register.
Let's say I have 4 shift registers and I want to send them these values:
B01111011, B01100000 ,B01011101 ,B01110101.
How do I choose which register will get which array?
I need a function to choose the register number and its array.

Simsso, 2020-08-13:
If you operate the shift registers in series you create a **single** shift register instance 
```cpp
int numberOfShiftRegisters = 4; // number of shift registers attached in series
int serialDataPin = 0; // DS
int clockPin = 1; // SHCP
int latchPin = 2; // STCP
ShiftRegister74HC595<numberOfShiftRegisters> sr(serialDataPin, clockPin, latchPin);
```
Then you set the values and the order in the `pinValues` array determines which of the four shift registers will output which byte.
```cpp
uint8_t pinValues[] = { B01111011, B01100000, B01011101, B01110101 };
sr.setAll(pinValues);
```

Please let me know whether that helps and close the issue, if it resolved the matter at hand.

yuvalkedar, 2020-08-13:
Thanks, I did just that yesterday and it works fine.

closed 2020-08-13 by Simsso