Simsso/ShiftRegister74HC595 #23: I made a Raspberry Pi Pico port of your library!

Devnol, 2021-07-15:
Hello, I wanted to use a 74HC595 display on one of my Raspberry Pi Pico projects and found this Arduino library. I have adapted it to work with it and have uploaded to GitHub at https://github.com/Devnol/ShiftRegister74HC595-Pico. I always try to show the original sources of my projects and I did as best as I could in showing that my library is a port of yours. However, I am not sure whether I have handled licensing correctly. Therefore, I would like you to please verify that my library correctly shows that it's a port of yours both inside LICENSE and the source files. Would it be easy for you to check out whether I have done so properly and maybe comment what I need to change so that my port abides by the MIT LICENSE correctly? 
Thanks in advance, Devnol.

Simsso, 2021-07-26:
Looks good to me! Thanks for your comment and for giving me credit in the fork.

On a separate note: I think it would be helpful to name the differences between this repo and your fork in your repo's `README.md`. Please close this issue, unless you have further questions. Cheers! :-)

Devnol, 2021-07-26:
Thank you so much for your time. I don't think there's any difference between my version of the library and yours in terms of end user functions, iirc it's a 1:1 map, only changing arduino.h api calls internally to pico stdlib functions. The readme is definitely a little rough on the edges and I do plan to work on it at one point. If there is any difference I have missed feel free to point it out. For now I  think closing the issue seems like the thing to do.
Cheers again and have a great summer break!

closed 2021-07-26 by Devnol