Simsso/ShiftRegister74HC595 #24: Shift register with millis funtion

itsboo07, 2021-07-23:
i want to use millis funtions on shift register outputs, im used sensors to tigger  lights but as im usig delay to turn on lights one by one. which makes the other sensor trigger useless...and this is sensitive to my project because it has people count . soo sensors has to work all time . 
soo how to use millis in shift register output pins

sfranzyshen, 2021-07-23:
> i want to use millis funtions on shift register outputs, im used sensors to tigger lights but as im usig delay to turn on lights one by one. which makes the other sensor trigger useless...and this [...]

Hi @itsboo07 ... why don't you post your code and we will try to help you with eliminating delays from your code. This issue isn't really related to this project, as delays are not used within this code base ... but I'll help you figure it out ... but I'll need a little more info to make that happen 😄

itsboo07, 2021-07-23:
sorry for posting here.
Im making a project of automatic stair case light with people count using 4 ir sensors and have used shift register to take the outputs.
the real problem i face is...i have set code in such a what that lights turn one other  one and goes off one other one. and have used delays for that...as my main motive of this project is about counting people in and out to trigger the funtions
real probelm is..when  one ir sensor is triggers and lights start to turn one by one.  and becuse of use delays . count funtions dont work  untill that funtion is comleted which makes a flaw .
 it would be really great if you help me as im new to all this and was real hard time building this whole code soo far.
 code:
`#include <ShiftRegister74HC595.h>
// create a global shift register object
// parameters: <number of shift registers> (data pin, clock pin, latch pin)
ShiftRegister74HC595<2> sr(14, 16, 15);

#define sensorPin1 5        //ir sensor 1
#define sensorPin2 7        //ir sensor 2
#define sensorPin3 9        //ir sensor 3
#define sensorPin4 11       //ir sensor 4

int sensorState1 = 0;
int sensorState2 = 0;
int sensorState3 = 0;
int sensorState4 = 0;
int count=0;
int time1=200;

void toupon() {                  //Turn on 1st to last light   
   for(int i=0;i<16;i++) {
      sr.set(i, HIGH);
      delay(time1);
      Serial.print("turn on light");
      Serial.println(i);
   }
}

void toupoff() {                  //Turn off from 1st to last light
   for(int i=0;i<16;i++) {
      sr.set(i, LOW);
      delay(time1);
      Serial.print("turn off light ");
      Serial.println(i);
   }
}

void todownon() {                 //turn on last to first light  
   
   for(int i = 15; i>=0; i--) {
      sr.set(i, HIGH);
      delay(time1);
      Serial.print("turn on light ");
      Serial.println(i);
   }
}
 
void todownoff() {                //turn off last to first light       

   for(int i = 15; i>=0; i--) {
      sr.set(i, LOW);
      delay(time1);
      Serial.print("turn off light ");
      Serial.println(i);
   }
}

void setup()
{
  Serial.begin(9600);
  Serial.println("FUN Started");
  Serial.print(count);
  pinMode (sensorPin1,INPUT);
  pinMode (sensorPin2, INPUT);
  pinMode (sensorPin3,INPUT);
  pinMode (sensorPin4,INPUT);
  
  
 
}

void loop() {
  int timex=800;
  
  sensorState1 = digitalRead(sensorPin1);
  sensorState2 = digitalRead(sensorPin2);
  sensorState3 = digitalRead(sensorPin3);
  sensorState4 = digitalRead(sensorPin4);
  int lower = digitalRead(sensorPin1);
  int upper = digitalRead(sensorPin4);

 if (count < 0)
    {
      count = 0;
    }
if(sensorState1 == LOW || sensorState4 == LOW ){   //counts people IN
    count++; 
    Serial.println(count);
    delay(timex);
  }

  if(sensorState2 == LOW || sensorState3 == LOW ){   //counts people OUT
    count--; 
    Serial.println(count);
    delay(timex);
  }
    if(count==0 && sensorState2 == LOW)
   
  {
    todownoff();
   
  }

if(count==0 && sensorState3 == LOW)
   
  {
    toupoff();
   
  }

 if (count==1  &&  sensorState1 == LOW)
 { 
  toupon();
  }
 if (count==1 &&  sensorState4 == LOW)
 { 
  todownon();
  } 

}
`

sfranzyshen, 2021-07-23:
OK, I may have over thunk this but I wanted to stay to exactly what you had ...
```
#include <ShiftRegister74HC595.h>

// create a global shift register object
// parameters: <number of shift registers> (data pin, clock pin, latch pin)

ShiftRegister74HC595<2> sr(14, 16, 15);

#define sensorPin1 5  // ir sensor 1
#define sensorPin2 7  // ir sensor 2
#define sensorPin3 9  // ir sensor 3
#define sensorPin4 11 // ir sensor 4

int sensorState1 = 0;
int sensorState2 = 0;
int sensorState3 = 0;
int sensorState4 = 0;

int count = 0;   // current counter count 

int time1 = 200; // delay between shiftregister updates
int timex = 800; // delay between counter updates 

unsigned long previousMode;    // Previous Millis for mode timer
unsigned long previousCount;   // Previous Millis for counter timer (delay between counts)
int           current = 0;     // current shiftregister
int           mode = 0;        // current mode 
bool          active = false;  // is mode active ?

void toupon() {  // Turn on 1st to last light   
  sr.set(current, HIGH);
  Serial.print("turn on light");
  Serial.println(current);
  current++;
  if(current > 15) {
    current = 0;
    active = false; // mode finished
  }
}

void toupoff() {  // Turn off from 1st to last light
  sr.set(current, LOW);
  Serial.print("turn off light ");
  Serial.println(current);
  current++;
  if(current > 15) {
    current = 0;
    active = false; // mode finished
  }
}

void todownon() {  // turn on last to first light  
  sr.set(current, HIGH);
  Serial.print("turn on light ");
  Serial.println(current);
  current--;
  if(current < 0) {
    current = 15;
    active = false; // mode finished
  }
}
 
void todownoff() {  // turn off last to first light       
  sr.set(current, LOW);
  Serial.print("turn off light ");
  Serial.println(current);
  current--;
  if(current < 0) {
    current = 15;
    active = false; // mode finished
  }
}

void setup() {
  Serial.begin(9600);
  Serial.println("FUN Started");
  
  pinMode(sensorPin1, INPUT);
  pinMode(sensorPin2, INPUT);
  pinMode(sensorPin3, INPUT);
  pinMode(sensorPin4, INPUT);
  
  previousMode = millis();  // init previousMode timer
  previousCount = millis(); // init previousCount timer 
}

void loop() {
  unsigned long currentMillis = millis(); // get current Millis for this frame

  sensorState1 = digitalRead(sensorPin1);
  sensorState2 = digitalRead(sensorPin2);
  sensorState3 = digitalRead(sensorPin3);
  sensorState4 = digitalRead(sensorPin4);

  if((currentMillis - previousCount) >= timex) { // delay between taking a reading
    previousCount = currentMillis;
    if(sensorState1 == LOW || sensorState4 == LOW ) {  // counts people IN
      count++; 
      Serial.println(count);
    }
    if(sensorState2 == LOW || sensorState3 == LOW ) {  // counts people OUT
      count--; 
      if(count < 0) { // bottom out counter count to zero
        count = 0;
      }
      Serial.println(count);
    }
    if(count == 0 && sensorState2 == LOW) {
      mode = 0;
      active = true;
    }
    if(count == 0 && sensorState3 == LOW) {
      mode = 1;
      active = true;
    }
    if(count == 1 && sensorState1 == LOW) {
      mode = 2;
      active = true;
    }
    if(count == 1 && sensorState4 == LOW) { 
      mode = 3;
      active = true;
    }
  }

  if(active) {
    if((currentMillis - previousMode) >= time1) { // delay between sr updates 
      previousMode = currentMillis;
      switch (mode) {
        case 0:
            todownoff();
        break;
        case 1:
            toupoff();
        break;
        case 2:
            toupon();
        break;
        case 3:
            todownon();
        break;
      }
    }
  }
}
```
hope that helped ... although I can see a problem when the counter switches between in and out it will screw up the mode mid processing and probably not giving you the expected results ...

itsboo07, 2021-07-23:
thanks for your time , but this code isnt funtioning.
its not turnings the lights on or off not sure whats wrong here

sfranzyshen, 2021-07-23:
what do you get at the serial console ... does it recognize the changes at the sensors?

itsboo07, 2021-07-23:
yes but not exactly
the issues are 
light turn on counts are runnning in loop and going just going all way from 16 ,17,18to 100...110..etc
and when sensor 4 or 3 are triggered its not working that way
and somtimes sensors are triggered funtions take some delay to start and also most of the time lights animiations are either going wrong or just not working
here is the video link on how my code performes
https://photos.app.goo.gl/Wjpon4ukS9M82Pqg8

sfranzyshen, 2021-07-23:
I made a change to the file above to initialize the current position ... can you see if that makes any difference? ... the problem also maybe the way timex is being applied ...

itsboo07, 2021-07-23:
hi...i would say the last code was better then this updated one.
before the code was updated. sensor 1 , 2 and 3 was working fine in last code
only sensor 4 was funtioning as same as sensor 3
also when sensor 1 and 2 are triggered together it goes to end less loop
the recent code has issues ..sensor 1 is just turning on the 15th light alone and isnt performing .
i would say code before updated works better then the last one

itsboo07, 2021-07-23:
is it possible for you to come in discord soo i can share my screen and you can watch the serial monitor yourself ?that could really be helpful .

sfranzyshen, 2021-07-23:
last try ... I left the delays in that were used during the sensor reading ... this will block the lights ... but is closer to your original code ...

```
#include <ShiftRegister74HC595.h>

// create a global shift register object
// parameters: <number of shift registers> (data pin, clock pin, latch pin)

ShiftRegister74HC595<2> sr(14, 16, 15);

#define sensorPin1 5  // ir sensor 1
#define sensorPin2 7  // ir sensor 2
#define sensorPin3 9  // ir sensor 3
#define sensorPin4 11 // ir sensor 4

int sensorState1 = 0;
int sensorState2 = 0;
int sensorState3 = 0;
int sensorState4 = 0;

int count = 0;   // current counter count 

int time1 = 200; // delay between shiftregister updates
int timex = 800; // delay between counter updates 

unsigned long previousMode;    // Previous Millis for mode timer
unsigned long previousCount;   // Previous Millis for counter timer (delay between counts)
int           current = 0;     // current shiftregister
int           mode = 0;        // current mode 
bool          active = false;  // is mode active ?

void toupon() {  // Turn on 1st to last light   
  sr.set(current, HIGH);
  Serial.print("turn on light");
  Serial.println(current);
  current++;
  if(current > 15) {
    current = 0;
    active = false; // mode finished
  }
}

void toupoff() {  // Turn off from 1st to last light
  sr.set(current, LOW);
  Serial.print("turn off light ");
  Serial.println(current);
  current++;
  if(current > 15) {
    current = 0;
    active = false; // mode finished
  }
}

void todownon() {  // turn on last to first light  
  sr.set(current, HIGH);
  Serial.print("turn on light ");
  Serial.println(current);
  current--;
  if(current < 0) {
    current = 15;
    active = false; // mode finished
  }
}
 
void todownoff() {  // turn off last to first light       
  sr.set(current, LOW);
  Serial.print("turn off light ");
  Serial.println(current);
  current--;
  if(current < 0) {
    current = 15;
    active = false; // mode finished
  }
}

void setup() {
  Serial.begin(9600);
  Serial.println("FUN Started");
  
  pinMode(sensorPin1, INPUT);
  pinMode(sensorPin2, INPUT);
  pinMode(sensorPin3, INPUT);
  pinMode(sensorPin4, INPUT);
  
  previousMode = millis();  // init previousMode timer
  previousCount = millis(); // init previousCount timer 
}

void loop() {
  unsigned long currentMillis = millis(); // get current Millis for this frame

  sensorState1 = digitalRead(sensorPin1);
  sensorState2 = digitalRead(sensorPin2);
  sensorState3 = digitalRead(sensorPin3);
  sensorState4 = digitalRead(sensorPin4);

  if(active) {
    if((currentMillis - previousMode) >= time1) { // delay between sr updates 
      previousMode = currentMillis;
      switch (mode) {
        case 0:
            todownoff();
        break;
        case 1:
            toupoff();
        break;
        case 2:
            toupon();
        break;
        case 3:
            todownon();
        break;
      }
    }
  }

  if(sensorState1 == LOW || sensorState4 == LOW ) {  // counts people IN
    count++; 
    Serial.println(count);
    delay(timex);
  }

  if(sensorState2 == LOW || sensorState3 == LOW ) {  // counts people OUT
    count--; 
    if(count < 0) { // bottom out counter count to zero
      count = 0;
    }
    Serial.println(count);
    delay(timex);
  }

  if(count == 0 && sensorState2 == LOW) {
    mode = 0;
    current = 0;
    active = true;
  }

  if(count == 0 && sensorState3 == LOW) {
    mode = 1;
    current = 0;
    active = true;
  }

  if(count == 1 && sensorState1 == LOW) {
    mode = 2;
    current = 15;
    active = true;
  }

  if(count == 1 && sensorState4 == LOW) { 
    mode = 3;
    current = 15;
    active = true;
  }
}

```

itsboo07, 2021-07-23:
uh...same issue sfranzyshen 
senssor 4 works senor 1, 2 , 3 isnt working the right way
if sensor 4 is triggered for like 2 time it just starts to count++ even its not triggered
it would be helpful if you seee the serial monitor yourself. i can share my screen in discord

sfranzyshen, 2021-07-23:
can you tell me more about your sensors ... how do they behave ...

* do they go HIGH or LOW when triggered
* are they real-time response (do they have a delay or no delay when changing state)

itsboo07, 2021-07-23:
When triggered SENSORS go LOW
Only sensor 4 works with trigger.. Yes it turned on lights one by one with some delay.

sfranzyshen, 2021-07-24:
this is as close to your original code as I can get ... it doesn't address your original problem ... yet it should run exactly (blocking) as your code does ... 

```
#include <ShiftRegister74HC595.h>

// create a global shift register object
// parameters: <number of shift registers> (data pin, clock pin, latch pin)

ShiftRegister74HC595<2> sr(14, 16, 15);

#define sensorPin1 5  // ir sensor 1
#define sensorPin2 7  // ir sensor 2
#define sensorPin3 9  // ir sensor 3
#define sensorPin4 11 // ir sensor 4

int sensorState1 = 0;
int sensorState2 = 0;
int sensorState3 = 0;
int sensorState4 = 0;

int count = 0;   // current counter count 

int time1 = 200; // delay between shiftregister updates
int timex = 800; // delay between counter updates 

unsigned long previousMode;    // Previous Millis for mode timer
unsigned long previousCount;   // Previous Millis for counter timer (delay between counts)
int           current = 0;     // current shiftregister
int           mode = 0;        // current mode 
bool          active = false;  // is mode active ?

void toupon() {  // Turn on 1st to last light   
  sr.set(current, HIGH);
  Serial.print("turn on light");
  Serial.println(current);
  current++;
  if(current > 15) {
    current = 0;
    active = false; // mode finished
  }
}

void toupoff() {  // Turn off from 1st to last light
  sr.set(current, LOW);
  Serial.print("turn off light ");
  Serial.println(current);
  current++;
  if(current > 15) {
    current = 0;
    active = false; // mode finished
  }
}

void todownon() {  // turn on last to first light  
  sr.set(current, HIGH);
  Serial.print("turn on light ");
  Serial.println(current);
  current--;
  if(current < 0) {
    current = 15;
    active = false; // mode finished
  }
}
 
void todownoff() {  // turn off last to first light       
  sr.set(current, LOW);
  Serial.print("turn off light ");
  Serial.println(current);
  current--;
  if(current < 0) {
    current = 15;
    active = false; // mode finished
  }
}

void setup() {
  Serial.begin(9600);
  Serial.println("FUN Started");
  
  pinMode(sensorPin1, INPUT);
  pinMode(sensorPin2, INPUT);
  pinMode(sensorPin3, INPUT);
  pinMode(sensorPin4, INPUT);
  
  previousMode = millis();  // init previousMode timer
  previousCount = millis(); // init previousCount timer 
}

void loop() {
  unsigned long currentMillis = millis(); // get current Millis for this frame

  sensorState1 = digitalRead(sensorPin1);
  sensorState2 = digitalRead(sensorPin2);
  sensorState3 = digitalRead(sensorPin3);
  sensorState4 = digitalRead(sensorPin4);

  if(active) {
    if((currentMillis - previousMode) >= time1) { // delay between sr updates 
      previousMode = currentMillis;
      switch (mode) {
        case 0:
            todownoff();
        break;
        case 1:
            toupoff();
        break;
        case 2:
            toupon();
        break;
        case 3:
            todownon();
        break;
      }
    }
    return;
  }
  
  if(sensorState1 == LOW || sensorState4 == LOW ) {  // counts people IN
    count++; 
    Serial.println(count);
    delay(timex);
  }

  if(sensorState2 == LOW || sensorState3 == LOW ) {  // counts people OUT
    count--; 
    if(count < 0) { // bottom out counter count to zero
      count = 0;
    }
    Serial.println(count);
    delay(timex);
  }
  
  if(count == 0 && sensorState2 == LOW) {
    mode = 0;
    current = 0;
    active = true;
  }
  
  if(count == 0 && sensorState3 == LOW) {
    mode = 1;
    current = 0;
    active = true;
  }
  
  if(count == 1 && sensorState1 == LOW) {
    mode = 2;
    current = 15;
    active = true;
  }
  
  if(count == 1 && sensorState4 == LOW) { 
    mode = 3;
    current = 15;
    active = true;
  }
}

```

sfranzyshen, 2021-07-24:
if that didn't work like your previous code did then I can only assume that your sensors have a reset timeout ... so when they get triggered ... they stay low for a period of time even if they are no longer getting triggered ...

itsboo07, 2021-07-24:
hi...i tried out the last code
serial monitor shows,
when sensor 2 is activated: turn off light 0
when sensor 1 is activated: turn on light15
but sensor 3 and 4 is working fine.

itsboo07, 2021-07-24:
hey thanks for your efforts on  my code. i really appreciate you 
I made this working by doing this, which is almost same to your code
my working code 
https://pastebin.com/iyrekAKe

itsboo07, 2021-07-24:
in this code?
https://pastebin.com/wRJsLVPx
can you eliminate  delays in count part..?
as the counter is working all fine/

closed 2021-07-25 by itsboo07