Simsso/ShiftRegister74HC595 #25: issue with shift register for 7segment disply

itsboo07, 2021-08-01:
hi im using 3 shift registers...and from that im using 1st and 2nd shift register for controlling 16 leds and i want to control the 7 segment display with the 3rd shift register...is it possible to do that? with this library without affecting the leds funtions?

sfranzyshen, 2021-08-01:
Hi again :smile: 
this driver (Arduino library) does not provide any direct support for seven segment displays ... but can be used to drive a seven segment display easily ... with that said ... how to do it ... 

first the physical connections ... I suggest looking at a few examples of using a seven segment display with a shift register to get an idea how the display is connected and programmed ... [here is a example for that](https://create.arduino.cc/projecthub/meljr/7-segment-led-displays-102-using-a-shift-register-6b6976) although this example does not use this driver ... rather it directly communicates with the shift register using shiftout ... however it does give you a good reference for the hardware setup ...

next your code will  need to update the display based on your counter to correctly display the number ... a table can be created using an array to handle the conversion ... looking at the source code of the example above, you will see such a table ...
```
byte symbol, symbols[] = {
B11111100, // 0
B01100000, // 1
B11011010, // 2
B11110010, // 3
B01100110, // 4
B10110110, // 5
B10111110, // 6
B11100000, // 7
B11111110, // 8
B11110110, // 9
B11101110, // A
B00111110, // B
B10011100, // C
B01111010, // D
B10011110, // E
B10001110 // F
};
```
This table is based on binary values and does not match your current maner of writing to the ShiftRegister74HC595 driver ... you can either use this data to create your own table ... or change the way you are writing to the ShiftRegister74HC595 driver ... 

right now you are calling sr.set() for the stair lights ... you might consider using sr.setNoUpdate() and sr.updateRegisters() to batch up each frame and write them all together ... and you could also do the same with the seven segment display at the same time ... 

I have been watching your progress since you opened your last issue ... and I'm curious to see how things turn out ... stay with it ... you will get it :smile:

itsboo07, 2021-08-01:
hi thanks for your fast response...let me explain my issue . im  making a staircase automatic light with four ultrasonic sensor to count person and with the person 1or 1+ it will turn on lights with the animations based on which side sensor is triggered ...i m using 16 leds from 2 shift register outputs now i want to add one more shift register for 7 segment display to show the number of persons...but not sure how is that possible..i can send you my code..
code i want to use for 7segment display
`byte bits;
bool decPt = true; // decimal point display flag
byte myfnNumToBits(int someNumber) {
 switch (someNumber) {
 case 0:
 return B11111100;
 break;
 case 1:
 return B01100000;
 break;
 case 2:
 return B11011010;
 break;
 case 3:
 return B11110010;
 break;
 case 4:
 return B01100110;
 break;
 case 5:
 return B10110110;
 break;
 case 6:
 return B10111110;
 break;
 case 7:
 return B11100000;
 break;
 case 8:
 return B11111110;
 break;
 case 9:
 return B11110110;
 break; 
 default:
 return B10010010; // Error condition, displays three vertical bars
 break; 
 }
}
void myfnUpdateDisplay(byte eightBits) {
 if (common == 'a') { // using a common anonde display?
 eightBits = eightBits ^ B11111111; // then flip all bits using XOR 
 }
 digitalWrite(latchPin, LOW); // prepare shift register for data
 shiftOut(dataPin, clockPin, LSBFIRST, eightBits); // send data
 digitalWrite(latchPin, HIGH); // update display
}`

also this is my person project code 

`#include <EEPROM.h>   //We need this library
#include <Arduino.h>  // for type definitions

#include <ShiftRegister74HC595.h>
#include "HCSR04_Custom.h"

//#define D_THRESHOLD 100     // the minimum maximum distance to indicate a trigger
#define MEASURE_COUNT 3;   // the number of measurement below the threshold in a row required to indicate a person

#define buttonPin 12
// lower group
#define e_sA 2 //echo pin
#define t_sA 3 //Trigger pin
#define e_sB 4 //echo pin
#define t_sB 5 //Trigger pin
// upper group
#define e_sC 6 //echo pin
#define t_sC 7 //Trigger pin
[... 416 lines cut ...]
    lastButtonState = buttonState;
  }

  if (buttonPushCounter == 1) {
    sr.setAll(pinValues1);
  }
  if (buttonPushCounter == 2) {
     sr.setAllHigh();     
  }
  if (buttonPushCounter == 3) {
   sr.setAllLow();
  }
  if (buttonPushCounter == 4) {
    
    sensor_control = true;
    //peopleCount();
    //animate_state();
  }
  if (buttonPushCounter > 4)
      {
       buttonPushCounter = 1;
      }

}`
it would be great if you can help me add a 7 segment disply in my peoject and im using common anode 7 segement display.....

itsboo07, 2021-08-01:
@sfranzyshen I have updated the above code... Can you please check... And help me how to add the 7 segment display? Sorry if I'm troubling you.
And I want the segment display to only work on button push counter 4 where sensors and counts work... In other button push counters its not related with any count or animations.

sfranzyshen, 2021-08-01:
Hi! Yes, I have forked your code and I'm working on helping you work the third shift register and seven segment display all working in conjunction with the rest of your code ... I should have something here soon :smile:

itsboo07, 2021-08-01:
> Hi! Yes, I have forked your code and I'm working on helping you work the third shift register and seven segment display all working in conjunction with the rest of your code ... I should have something here soon 😄

itsboo07, 2021-08-01:
Sorry it was by mistake....

sfranzyshen, 2021-08-02:
after looking over you code I came up with a way to handle this without changing your original code ... I used a multidimensional array to hold each digit from 0 to 9 and A, B,C, D, E, and a ERROR pattern (three lines) that can be used in a for loop to update the third shift register without changing the way your code writes to the shift registers ... I have not tested compiling this code ... and of course I didn't run it since I don't have your setup ...

https://gist.github.com/sfranzyshen/db79ae57beb6742bc6fc3a125f84ce08/revisions

itsboo07, 2021-08-02:
Thanks @sfranzyshen. I will try to check and Compile. And also I'm using Common anode... Soo do I have to change any stuffs for common anode?

sfranzyshen, 2021-08-02:
> Thanks @sfranzyshen. I will try to check and Compile. And also I'm using Common anode... Soo do I have to change any stuffs for common anode?

in the code I create two versions of the Digits[][] multidimensional array ... one for the common anode and one for common cathode ... you would REM (//) out the one not used ... currently it's set for common anode ... basicly they are inverted (0 = 1 and 1 = 0) of each other ...

itsboo07, 2021-08-02:
> in the code I create two versions of the Digits[][] multidimensional array ... one for the common anode and one for common cathode ... you would REM (//) out the one not used ... currently it's set [...]

Sorry my bad... I was trying to read code in my mobile.
What about Sr.setAllpinValues instead of srsetAllHigh or Low?

sfranzyshen, 2021-08-02:
> Thanks @sfranzyshen. I will try to check and Compile. And also I'm using Common anode... Soo do I have to change any stuffs for common anode?
> And also what if I use sr.setAllpinValues instead of setAllHigh or setAllLow? Like this
> uint8_t pinValues0[] = { B00000000, B00000000};
> Soo this could only update the 1st two shift registers right? Without disturbing the 3rd one... Would this work like this?

I haven't tried but I believe it will work without passing the third byte, but I expect it will zero it out either way ... so you should call set_counter() after calling any of the setAll functions ... the display will more than flicker when this happens but should be acceptable ...

itsboo07, 2021-08-02:
> > Thanks @sfranzyshen. I will try to check and Compile. And also I'm using Common anode... Soo do I have to change any stuffs for common anode?
> > And also what if I use sr.setAllpinValues instead of setAllHigh or setAllLow? Like this
> > uint8_t pinValues0[] = { B00000000, B00000000};
> > Soo this could only update the 1st two shift registers right? Without disturbing the 3rd one... Would this work like this?
> 
> I haven't tried but I believe it will work without passing the third byte, but I expect it will zero it out either way but you should try ... if it disturbs the third ... you should call set_counter() after calling any of the setAll functions ... the display will more than likely flicker when this happens but should be acceptable ... In fact you might want to change the way I'm calling it in the detect_button() ... maybe un-necessarily calling it too many times ...

Yeah.... You have called under animate_state which does only animations  for person count 0 and 1. And for person count 1+ their is no animations. 
I think they should be placed somewhere separate or may be under person= person+1
Soo it changes accordingly

itsboo07, 2021-08-02:
hi i tried but it didnot work...but for now thats not a issue..issue i have now is...my schematic and code working fine with leds as output...but if i try that with relay its not working..not sure why..can you help whats going wrong here

itsboo07, 2021-08-02:
also  this the is the schematic i used to connect to relay. instead of relay if i use led it works..but relays dosnt work also i set to LOW.so relays get triggered with LOW not HIGH 
this the the relay module im using
https://www.electronicscomp.com/8-channel-5v-relay-module-with-optocoupler?gclid=CjwKCAjwr56IBhAvEiwA1fuqGi6MQScXaBy7GMNkRiPjn5MLaUD8nLvqXVCccloAFSfkz9ENIY2hJBoC6JkQAvD_BwE
[image]

itsboo07, 2021-08-02:
hey @sfranzyshen  with few changes 7 segment works..but it keeps flickering and leds are glowing really very low.

itsboo07, 2021-08-02:
@sfranzyshen also its turning off 8 lights from shift register 1 not sure why

sfranzyshen, 2021-08-03:
> hey @sfranzyshen with few changes 7 segment works..but it keeps flickering .
great to hear ... lets see your code ... either update your old gist above or post a new link ...

itsboo07, 2021-08-03:
> > hey @sfranzyshen with few changes 7 segment works..but it keeps flickering .
> 
> great to hear ... lets see your code ... either update your old gist above or post a new link ...

Sure... And also for each button push counter it changes a b c d. And in sensor mode it it shifts with numerical digits to count.
Thanks for your help.

itsboo07, 2021-08-03:
@sfranzyshen I need a help... Eveything works good... Execpt for one issue... When in button state 4 even tho eveything is SETHIGH which supposed turn off all relay.. But few are turned on... Why is that so...?

itsboo07, 2021-08-03:
> > hey @sfranzyshen with few changes 7 segment works..but it keeps flickering .
> 
> great to hear ... lets see your code ... either update your old gist above or post a new link ...

https://gist.github.com/itsboo07/5e7f63ca8db4a4043789a31577d9f379
hey!! let me know when you check and update me if I should fix any part of the code. also I want to make this code private for now.

sfranzyshen, 2021-08-03:
I have forked your code ... if you want to keep this private you can edit these messages and remove the links ... I have a full schedule today but I will find time to review your code at some point (today), and see if I can help solve the flickering problem ...

sfranzyshen, 2021-08-03:
> also eveything works soo smooth and perfect to me soo far...is it necessary to use bypass capacitors with shift register?

that really depends on your power source and number of shift registers used ... however, [de-coupling](http://www.thebox.myzen.co.uk/Tutorial/De-coupling.html) can cause all types of intermittent problems ... so for the cost of a capacitor per shift register ... it's a good practice to use ... also there is not downside to using them :smile:

itsboo07, 2021-08-03:
> I have forked your code ... if you want to keep this private you can edit these messages and remove the links ... I have a full schedule today but I will find time to review your code at some point (today), and see if I can help solve the flickering problem ...

thanks and forgot to tell you that flickering issue is fixed

itsboo07, 2021-08-03:
@sfranzyshen as you know im using 3 shift resgisters soo if i have to use a capacitor should i add one for each SR or elsewhere.

sfranzyshen, 2021-08-03:
> thanks and forgot to tell you that flickering issue is fixed

That's great! :smile:

> @sfranzyshen as you know im using 3 shift resgisters soo if i have to use a capacitor should i add one for each SR or elsewhere.

one cap (0.1 uF) for [each shift register across the power and ground](https://i.stack.imgur.com/1XiV1.png) 

[image]

itsboo07, 2021-08-03:
Adding capacitor this way would work? 
[image]

sfranzyshen, 2021-08-03:
that is another issue ... that is a cap across the latch pin ... see [this post](https://arduino.stackexchange.com/questions/39827/is-the-shiftout-tutorial-on-the-official-arduino-website-really-wrong-regarding) for more information ... this can also be used ... in addition to the ceramic caps at each shift register ... you might be fine without using any caps at all ... it's up to you if you develop problems to which method to use ... both ... one or the other ... or none ... experiment and see what works

itsboo07, 2021-08-04:
Actually eveything works fine...except for one issue...
When EEPROM writes and read... It all. Works fine for all the button push counter expect for 4...thats for sensors and animations. 
What happens here is when in sensor mode.. If I unplug and replug the controller... EEPROM reads last button push and if it's sensor mode... That's push 4..then some of the relays gets turned on. Once the animation starts it all goes off and gets normal...
Even I set all values HIGH If EEPROM read is button 4. 
This only happens when put everything to high. If I use like just one output low and other high like
B011111111 then sometimes it works.. And sometimes instead one one relay it turns 2 or 3 relays..
Whats the issue going on here..? 
Without 7segment it works fine... But after adding 7 segment.. Their is some issue with this.

sfranzyshen, 2021-08-04:
depending on how you are powering everything you might be experiencing a current drop when initializing many relays simultaneously ... you might try powering the relay boards separately than the arduino and if you do decide to power the relay boards separately just remember to inter-connect power sources with a common ground ... just make sure you have adequate power supply not just voltage but also current size (amps)

itsboo07, 2021-08-04:
Yeah I'm using Two power source only. Still I have the issue.... And gnds are connected.... I don't have any issue in animations or all on or all off... Only during the initial powering of arduino... Have a problem... When I was using Two shift registers....it was working fine....

itsboo07, 2021-08-04:
hey!! this is my complete project schematic(link below) @sfranzyshen 
please can you check and update me if i need to fix anything.
also I added 0.1uF capacitors for each shift register.
https://drive.google.com/file/d/1Z-hPnSrhU-NGdNno1RJHvG6AQrOjewd4/view?usp=sharing

sfranzyshen, 2021-08-04:
> @sfranzyshen please can you check and update me if i need to fix anything.

looks clean to me :smile:

itsboo07, 2021-08-04:
hai thanks for your response ..is 7segement connected correctly?
I had flickering issues when connected with Arduino 5v soo instead I connected with external 5v supply and flickering stopped. and even leds looks brighter too. just worried did I make the connections correctly.

sfranzyshen, 2021-08-04:
> hai thanks for your response ..is 7segement connected correctly?
> I had flickering issues when connected with Arduino 5v soo instead I connected with external 5v supply and flickering stopped. and even leds looks brighter too. just worried did I make the connections correctly.

I would say you are exceeding the output of the Arduion's regulator. The 5V pin is limited to 500mA of current if you are powering the Arduino with a USB cable. The onboard voltage regulator is rated for 800mA, but due to power dissipation issues, you shouldn't go over 400 to 500mA.

itsboo07, 2021-08-04:
> > hai thanks for your response ..is 7segement connected correctly?
> > I had flickering issues when connected with Arduino 5v soo instead I connected with external 5v supply and flickering stopped. and even leds looks brighter too. just worried did I make the connections correctly.
> 
> I would say you are exceeding the output of the Arduino's voltage regulator. The 5V pin is limited to 500mA of current if you are powering the Arduino with a USB cable. The onboard voltage regulator is rated for 800mA, but due to power dissipation issues, you shouldn't go over 400 to 500mA.

I understand the issue 
thats why i connected the 7segment 5v to external 5v supply instead of Arduino 5v. and it fixed the flickering. 
also each ultrasonic sensor only take 2mA soo all four wont cross more than 10mA. apart from ultrasonic im using a touch module and a potentiometer soo all this wont eat that much amps i guess..and also arduino 5v is supplied to shift registers.
and i fixed the stating power issue by turning on 2 relays first and added some delay to it and then turn all off and it works.
i would really thank you for your support all this time mate.
i guess i can close this issue. and really love the way you respond and vouched me in making this project happen. thanks again.

sfranzyshen, 2021-08-04:
> i would really thank you for your support all this time mate.
> i guess i can close this issue. and really love the way you respond and vouched me in making this project happen. thanks again.

We are all here for each other ... once we all can remember that ... things will be good :smile: ~peace

closed 2021-08-04 by itsboo07