Simsso/ShiftRegister74HC595 #27: Library is not working with ultra sonic sensor

VarunKumaran, 2021-08-13:


VarunKumaran, 2021-08-13:
#include <ShiftRegister74HC595.h>
#define echoPin 4// attach pin D2 of Esp8266 NodeMcu to pin Echo of HC-SR04
#define trigPin 5//attach pin D1 of Esp8266 NodeMcu to pin Trig of HC-SR04
// defines variables
long duration; // variable for the duration of sound wave travel
int d; // variable for the distance measurement

ShiftRegister74HC595<1> sr(3, 1, 10);

void setup() {
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an OUTPUT
  pinMode(echoPin, INPUT); // Sets the echoPin as an INPUT
  Serial.begin(9600); // // Serial Communication is starting with 9600 of baudrate speed
}
void loop() {
  // Clears the trigPin condition
  digitalWrite(trigPin, LOW);
  delayMicroseconds(5);
  // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);
  // Calculating the distance
  d = (duration * 0.034 / 2)+ 3; // Speed of sound wave divided by 2 (go and back)
  // Displays the distance on the Serial Monitor
  Serial.print("Distance: ");
  Serial.print(d);
  Serial.println(" cm");
  delay(1000);
  
  if (d<25 || d>200){ 
    uint8_t pinValues[] = {B00011111};
    sr.setAll(pinValues);
    delay(1000);
  }
  else if(d<48 & d>=25){
    uint8_t pinValues[] = {B00011110};
    sr.setAll(pinValues);
    delay(500);
  }
  else if(d<72 & d>=48){
     uint8_t pinValues[] = {B00011100};
    sr.setAll(pinValues);
    delay(500);
  }
  else if(d<105 & d>=72){
    uint8_t pinValues[] = {B00011000};
    sr.setAll(pinValues);
    delay(500);
  }
  else if(d<200 & d>=105){
    uint8_t pinValues[] = {B00010000};
    sr.setAll(pinValues);
    delay(500);
  }
}

VarunKumaran, 2021-08-13:
This is My code. My ultrasonic sensor 's value are displayed in serial monitor and it is perfect. But the desired our for led is not coming. All I am getting is pin 0-7 in shift register is High,hence all the leds are light up and there is no change when distance (d) is changed.

VarunKumaran, 2021-08-13:
I have written this sample code for finding the level of water in our tank. Sooo kindly if anyone knows how to solve this plss respond. 
My Gmail acc is varunkumaran1234@gmail.com

sfranzyshen, 2021-08-13:
Hi @VarunKumaran 

It looks like the problem is with your if else statements are using the '&' Pointer Access Operator and not the '&&' logical 'AND' Operator ... change your code like this ...
```
	if (d < 25 || d > 200) { // OR
		uint8_t pinValues[] = {B00011111};
		sr.setAll(pinValues);
		delay(1000);
	}
	else if(d < 48 && d >= 25) { // AND
		uint8_t pinValues[] = {B00011110};
		sr.setAll(pinValues);
		delay(500);
	}
	else if(d < 72 && d >= 48) { // AND
		uint8_t pinValues[] = {B00011100};
		sr.setAll(pinValues);
		delay(500);
	}
	else if(d < 105 && d >= 72) { // AND
		uint8_t pinValues[] = {B00011000};
		sr.setAll(pinValues);
		delay(500);
	}
	else if(d < 200 && d >= 105) { // AND
		uint8_t pinValues[] = {B00010000};
		sr.setAll(pinValues);
		delay(500);
	}
```

VarunKumaran, 2021-08-13:
Okayy I will try this code

VarunKumaran, 2021-08-13:
No it's not working, after uploading the code all the pins become high and does not change with the condition statement

VarunKumaran, 2021-08-13:
HI @sfranzyshen  
Thank you soo much for looking into my issue but the problem I am facing is not solved yet. If you have any other ideas please share.
Thank you in advance.

VarunKumaran, 2021-08-13:
@Simsso @lathoub @per1234

sfranzyshen, 2021-08-13:
> No it's not working, after uploading the code all the pins become high and does not change with the condition statement

* you only changed the `'&'` to `'&&'` in your code ??
* nothing should have changed for the ultra sonic sensor code and should function exactly as it did before ... so if it's not ... something else is wrong
* could you post the serial output (what prints out for the value of `'d'`) ??

sfranzyshen, 2021-08-13:
the way your code is written is if the value of `d` (defaults to 0) is less than 25 "OR ||" if the value of `d` is greater than 200 the code will set the output to `B00011111` and will not perform any further test ... then it loops ... so if the value only falls inside this window nothing else will happen ... consider re-writing the first if statement (or even eliminate it)

sfranzyshen, 2021-08-13:
also try declaring `'d'` as a `long` rather than a `int`

`int d; // variable for the distance measurement
`
to
`long d; // variable for the distance measurement
`

VarunKumaran, 2021-08-13:
> * you only changed the `'&'` to `'&&'` in your code ??
> * nothing should have changed for the ultra sonic sensor code and should function exactly as it did before ... so if it's not ... something [...]

Yess I did exactly the same.. wait I will post my serial monitor outpur

sfranzyshen, 2021-08-13:
have you verified your shift registers setup function as expected?

VarunKumaran, 2021-08-13:
> have you verified your shift registers setup function as expected?

Yess I have and it works perfectly well with the example given.

VarunKumaran, 2021-08-13:
[image]
[image]

VarunKumaran, 2021-08-13:
As you can see I have Removed the OR || statement and changed my 'int' d to 'long d'

VarunKumaran, 2021-08-13:
[image]

This was my example code to check wether my connections are right. it works as expected.

sfranzyshen, 2021-08-13:
ok I may have over thunk this but ... try this ... and post the serial output ...

```
#include <ShiftRegister74HC595.h>
#define echoPin 4  // attach pin D2 of Esp8266 NodeMcu to pin Echo of HC-SR04
#define trigPin 5 // attach pin D1 of Esp8266 NodeMcu to pin Trig of HC-SR04

// defines variables
long duration;    // variable for the duration of sound wave travel
long d;         // variable for the distance measurement

ShiftRegister74HC595<1> sr(3, 1, 10);

void setup() {
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an OUTPUT
  pinMode(echoPin, INPUT);  // Sets the echoPin as an INPUT
  Serial.begin(9600);     // Serial Communication is starting with 9600 of baudrate speed
}

void loop() {

  uint8_t pinValues[] = {};
  
  // Clears the trigPin condition
  digitalWrite(trigPin, LOW);
  delayMicroseconds(5);

  // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);

  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);

  // Calculating the distance
  d = (duration * 0.034 / 2) + 3; // Speed of sound wave divided by 2 (go and back)

  // Displays the distance on the Serial Monitor
  Serial.print("Distance: ");
  Serial.print(d);
  Serial.println(" cm");
  delay(1000);

  if((int) d < 48 && (int) d >= 25) {
    pinValues[0] = {B00011110};
    sr.setAll(pinValues);
    Serial.println("mode 1");
    delay(500);
  }
  
  else if((int) d < 72 && (int) d >= 48) {
    pinValues[0] = {B00011100};
    sr.setAll(pinValues);
    Serial.println("mode 2");
    delay(500);
  }
  
  else if((int) d < 105 && (int) d >= 72) {
    pinValues[0] = {B00011000};
    sr.setAll(pinValues);
    Serial.println("mode 3");
    delay(500);
  }
  
  else if((int) d < 200 && (int) d >= 105) {
    pinValues[0] = {B00010000};
    sr.setAll(pinValues);
    Serial.println("mode 4");
    delay(500);
  }

  else if ((int) d < 25 || (int) d > 200) {
    pinValues[0] = {B00011111};
    sr.setAll(pinValues);
    Serial.println("mode 0");
    delay(1000);
  }
}
```

VarunKumaran, 2021-08-13:
[image]
In the serial output we get the values but the leds does not change at all

VarunKumaran, 2021-08-13:
> ok I may have over thunk this but ... try this ... and post the serial output ...
> 
> ```
> #include <ShiftRegister74HC595.h>
> #define echoPin 4  // attach pin D2 of Esp8266 NodeMcu to pin Echo of HC-SR04
> #define trigPin 5 // attach pin D1 of Esp8266 NodeMcu to pin Trig of HC-SR04
> 
> // defines variables
> long duration;    // variable for the duration of sound wave travel
> long d;           // variable for the distance measurement
> long last_d;      // track change
> 
> ShiftRegister74HC595<1> sr(3, 1, 10);
> 
> void setup() {
>   pinMode(trigPin, OUTPUT); // Sets the trigPin as an OUTPUT
>   pinMode(echoPin, INPUT);  // Sets the echoPin as an INPUT
>   Serial.begin(9600);     // Serial Communication is starting with 9600 of baudrate speed
> }
> 
> void loop() {
> 
>   uint8_t pinValues[] = {};
>   
>   // Clears the trigPin condition
>   digitalWrite(trigPin, LOW);
>   delayMicroseconds(5);
> 
>   // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
>   digitalWrite(trigPin, HIGH);
>   delayMicroseconds(10);
>   digitalWrite(trigPin, LOW);
> 
>   // Reads the echoPin, returns the sound wave travel time in microseconds
>   duration = pulseIn(echoPin, HIGH);
> 
>   // Calculating the distance
>   d = (duration * 0.034 / 2) + 3; // Speed of sound wave divided by 2 (go and back)
> 
>   // Displays the distance on the Serial Monitor
>   Serial.print("Distance: ");
>   Serial.print(d);
>   Serial.println(" cm");
>   delay(1000);
> 
>   if((int) d < 48 && (int) d >= 25) {
>     pinValues[0] = {B00011110};
>     Serial.println("mode 1");
>   }
>   
>   else if((int) d < 72 && (int) d >= 48) {
>     pinValues[0] = {B00011100};
>     Serial.println("mode 2");
>   }
>   
>   else if((int) d < 105 && (int) d >= 72) {
>     pinValues[0] = {B00011000};
>     Serial.println("mode 3");
>   }
>   
>   else if((int) d < 200 && (int) d >= 105) {
>     pinValues[0] = {B00010000};
>     Serial.println("mode 4");
>   }
> 
>   else if ((int) d < 25 || (int) d > 200) {
>     pinValues[0] = {B00011111};
>     Serial.println("mode 0");
>   }
> 
>   if(last_d != d) { // only update if different
>     last_d = d;
>     sr.setAll(pinValues);
>     delay(500);
>   }
> }
> ```
> 
> what I changed
> 
> * declare variable `d` as a long
> * cast variable `d` to int for comarison in the if functions
> * declared pinValues[] at the top of the loop function
> * moved the order of if else statements
> * added debug mode printing to serial

Tried this code

itsboo07, 2021-08-13:
can you try this @VarunKumaran 
`#include <ShiftRegister74HC595.h>
ShiftRegister74HC595<1> sr(3, 1, 10);

#define echoPin 4  // attach pin D2 of Esp8266 NodeMcu to pin Echo of HC-SR04
#define trigPin 5 // attach pin D1 of Esp8266 NodeMcu to pin Trig of HC-SR04

// defines variables
long duration;    // variable for the duration of sound wave travel
long d;           // variable for the distance measurement
long last_d;      // track change

 uint8_t  pinValues1[] = {B00011110};//mode1
 uint8_t  pinValues2[] = {B00011100};//mode2
 uint8_t  pinValues3[] = {B00011000};//mode3
 uint8_t  pinValues4[] = {B00010000};//mode4
 uint8_t  pinValues0[] = {B00011111};//mode0

void setup() {
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an OUTPUT
  pinMode(echoPin, INPUT);  // Sets the echoPin as an INPUT
  Serial.begin(9600);     // Serial Communication is starting with 9600 of baudrate speed
}

void loop() {
  
  // Clears the trigPin condition
  digitalWrite(trigPin, LOW);
  delayMicroseconds(5);

  // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);

  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);

  // Calculating the distance
  d = (duration * 0.034 / 2) + 3; // Speed of sound wave divided by 2 (go and back)

  // Displays the distance on the Serial Monitor
  Serial.print("Distance: ");
  Serial.print(d);
  Serial.println(" cm");
  delay(1000);

  if(last_d != d) {
    
  if((int) d < 48 && (int) d >= 25) { 
    //pinValues0[] = {B00011110};
    sr.setAll(pinValues1);
    Serial.println("mode 1");
    delay(500);
    last_d = d;
  }
  
  if((int) d < 72 && (int) d >= 48) {  
    //pinValues[0] = {B00011100};
    sr.setAll(pinValues2);
    Serial.println("mode 2");
    delay(500);
    last_d = d;
  }
  
  if((int) d < 105 && (int) d >= 72) {  
    //pinValues[0] = {B00011000};
    sr.setAll(pinValues3);
    Serial.println("mode 3");
    delay(500);
    last_d = d;
  }
  
  if((int) d < 200 && (int) d >= 105) {
    //pinValues[0] = {B00010000};
    sr.setAll(pinValues4);
    Serial.println("mode 4");
    delay(500);
    last_d = d;
  }

  if ((int) d < 25 || (int) d > 200) {
    //pinValues[0] = {B00011111};
    sr.setAll(pinValues0);
    Serial.println("mode 0");
    delay(500);
    last_d = d;
  }
 }
}`

sfranzyshen, 2021-08-13:
here is my latest try ...

```
#include <ShiftRegister74HC595.h>
#define echoPin 4  // attach pin D2 of Esp8266 NodeMcu to pin Echo of HC-SR04
#define trigPin 5 // attach pin D1 of Esp8266 NodeMcu to pin Trig of HC-SR04

// defines variables
long duration;    // variable for the duration of sound wave travel
long d;           // variable for the distance measurement
long last_d;      // track change

ShiftRegister74HC595<1> sr(3, 1, 10);

void setup() {
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an OUTPUT
  pinMode(echoPin, INPUT);  // Sets the echoPin as an INPUT
  Serial.begin(9600);     // Serial Communication is starting with 9600 of baudrate speed
}

void loop() {

  uint8_t pinValues[1];
  
  // Clears the trigPin condition
  digitalWrite(trigPin, LOW);
  delayMicroseconds(5);

  // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);

  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);

  // Calculating the distance
  d = (duration * 0.034 / 2) + 3; // Speed of sound wave divided by 2 (go and back)

  // Displays the distance on the Serial Monitor
  Serial.print("Distance: ");
  Serial.print(d);
  Serial.println(" cm");
  delay(1000);

  if((int) d < 48 && (int) d >= 25) {
    pinValues[0] = B00011110;
    Serial.println("mode 1");
  }
  
  else if((int) d < 72 && (int) d >= 48) {
    pinValues[0] = B00011100;
    Serial.println("mode 2");
  }
  
  else if((int) d < 105 && (int) d >= 72) {
    pinValues[0] = B00011000;
    Serial.println("mode 3");
  }
  
  else if((int) d < 200 && (int) d >= 105) {
    pinValues[0] = B00010000;
    Serial.println("mode 4");
  }

  else if ((int) d < 25 || (int) d > 200) {
    pinValues[0] = B00011111;
    Serial.println("mode 0");
  }

  if(last_d != d) { // only update if different
    last_d = d;
    sr.setAll(pinValues);
    delay(500);
  }
}
```

VarunKumaran, 2021-08-13:
I will try both the codes.. just give me 5 mins

itsboo07, 2021-08-13:
hey @sfranzyshen hope you doing good.....thought i could may be help...hope you don't take it wrong.

VarunKumaran, 2021-08-13:
> can you try this @VarunKumaran
> 
> ```
> #include <ShiftRegister74HC595.h>
> ShiftRegister74HC595<1> sr(3, 1, 10);
> 
> #define echoPin 4  // attach pin D2 of Esp8266 NodeMcu to pin Echo of HC-SR04
> #define trigPin 5 // attach pin D1 of Esp8266 NodeMcu to pin Trig of HC-SR04
> 
> // defines variables
> long duration;    // variable for the duration of sound wave travel
> long d;           // variable for the distance measurement
> long last_d;      // track change
> 
>  uint8_t  pinValues1[] = {B00011110};   //mode1
>  uint8_t  pinValues2[] = {B00011100};   //mode2
>  uint8_t  pinValues3[] = {B00011000};   //mode3
>  uint8_t  pinValues4[] = {B00010000};   //mode4
>  uint8_t  pinValues0[] = {B00011111};   //mode0
> 
> void setup() {
>   pinMode(trigPin, OUTPUT); // Sets the trigPin as an OUTPUT
>   pinMode(echoPin, INPUT);  // Sets the echoPin as an INPUT
>   Serial.begin(9600);     // Serial Communication is starting with 9600 of baudrate speed
> }
> 
> void loop() {
>   
>   // Clears the trigPin condition
>   digitalWrite(trigPin, LOW);
>   delayMicroseconds(5);
> 
>   // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
>   digitalWrite(trigPin, HIGH);
>   delayMicroseconds(10);
>   digitalWrite(trigPin, LOW);
> 
>   // Reads the echoPin, returns the sound wave travel time in microseconds
>   duration = pulseIn(echoPin, HIGH);
> 
>   // Calculating the distance
>   d = (duration * 0.034 / 2) + 3; // Speed of sound wave divided by 2 (go and back)
> 
>   // Displays the distance on the Serial Monitor
>   Serial.print("Distance: ");
>   Serial.print(d);
>   Serial.println(" cm");
>   delay(1000);
> 
>   if(last_d != d) { 
>     
>   if((int) d < 48 && (int) d >= 25) { 
>     //pinValues0[] = {B00011110};
>     sr.setAll(pinValues1);
>     Serial.println("mode 1");
>     last_d = d;
>     delay(500);
>   }
>   
>   if((int) d < 72 && (int) d >= 48) {  
>     //pinValues[0] = {B00011100};
>     sr.setAll(pinValues2);
>     Serial.println("mode 2");
>     last_d = d;
>     delay(500);
>   }
>   
>   if((int) d < 105 && (int) d >= 72) {  
>     //pinValues[0] = {B00011000};
>     sr.setAll(pinValues3);
>     Serial.println("mode 3");
>     last_d = d;
>     delay(500);
>   }
>   
>   if((int) d < 200 && (int) d >= 105) {
>     //pinValues[0] = {B00010000};
>     sr.setAll(pinValues4);
>     Serial.println("mode 4");
>     last_d = d;
>     delay(500);
>   }
> 
>   if ((int) d < 25 || (int) d > 200) {
>     //pinValues[0] = {B00011111};
>     sr.setAll(pinValues0);
>     Serial.println("mode 0");
>     last_d = d;
>     delay(500);
>   }
>  }
> }
> ```

[image]
Serial monitor output but still no change in the hardware part

sfranzyshen, 2021-08-13:
> hey @sfranzyshen hope you doing good.....thought i could may be help...hope you don't take it wrong.

the more the merrier :smile:

VarunKumaran, 2021-08-13:
> here is my latest try too ...
> 
> ```
> #include <ShiftRegister74HC595.h>
> #define echoPin 4  // attach pin D2 of Esp8266 NodeMcu to pin Echo of HC-SR04
> #define trigPin 5 // attach pin D1 of Esp8266 NodeMcu to pin Trig of HC-SR04
> 
> // defines variables
> long duration;    // variable for the duration of sound wave travel
> long d;           // variable for the distance measurement
> long last_d;      // track change
> 
> ShiftRegister74HC595<1> sr(3, 1, 10);
> 
> void setup() {
>   pinMode(trigPin, OUTPUT); // Sets the trigPin as an OUTPUT
>   pinMode(echoPin, INPUT);  // Sets the echoPin as an INPUT
>   Serial.begin(9600);     // Serial Communication is starting with 9600 of baudrate speed
> }
> 
> void loop() {
> 
>   uint8_t pinValues[1];
>   
>   // Clears the trigPin condition
>   digitalWrite(trigPin, LOW);
>   delayMicroseconds(5);
> 
>   // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
>   digitalWrite(trigPin, HIGH);
>   delayMicroseconds(10);
>   digitalWrite(trigPin, LOW);
> 
>   // Reads the echoPin, returns the sound wave travel time in microseconds
>   duration = pulseIn(echoPin, HIGH);
> 
>   // Calculating the distance
>   d = (duration * 0.034 / 2) + 3; // Speed of sound wave divided by 2 (go and back)
> 
>   // Displays the distance on the Serial Monitor
>   Serial.print("Distance: ");
>   Serial.print(d);
>   Serial.println(" cm");
>   delay(1000);
> 
>   if((int) d < 48 && (int) d >= 25) {
>     pinValues[0] = B00011110;
>     Serial.println("mode 1");
>   }
>   
>   else if((int) d < 72 && (int) d >= 48) {
>     pinValues[0] = B00011100;
>     Serial.println("mode 2");
>   }
>   
>   else if((int) d < 105 && (int) d >= 72) {
>     pinValues[0] = B00011000;
>     Serial.println("mode 3");
>   }
>   
>   else if((int) d < 200 && (int) d >= 105) {
>     pinValues[0] = B00010000;
>     Serial.println("mode 4");
>   }
> 
>   else if ((int) d < 25 || (int) d > 200) {
>     pinValues[0] = B00011111;
>     Serial.println("mode 0");
>   }
> 
>   if(last_d != d) { // only update if different
>     last_d = d;
>     sr.setAll(pinValues);
>     delay(500);
>   }
> }
> ```

[image]
still the same @sfranzyshen

VarunKumaran, 2021-08-13:
https://user-images.githubusercontent.com/86418940/129415851-3d09065c-657c-4c09-8fda-db30c0d41e3b.mp4

With my example code I am shared earlier^ the leds work as expected.

itsboo07, 2021-08-13:
@VarunKumaran i would suggest you try this library for sonic sensors https://www.arduinolibraries.info/libraries/hcsr04
and use flags to make it work. just making readings in ultrasonic sensors wont trigger a fun.

itsboo07, 2021-08-13:
@VarunKumaran share me your discord Id . may be i can try to help you their

sfranzyshen, 2021-08-13:
the problem seems to be with the writing to the shift register using the setAll() function ... if the example from this driver worked ... than one of our examples should have changed the state of the lights ... can you test using the test code again and verify it's still working ... and if so please post the actual code used to get a verified working test?

VarunKumaran, 2021-08-13:
> @VarunKumaran i would suggest you try this library for sonic sensors https://www.arduinolibraries.info/libraries/hcsr04
> and use flags to make it work. I think just making readings in ultrasonic sensors wont trigger a function.

Okay will try but I am using JSN-SR04T which is a water proof ultrasonic sensor instead of HCSR04. They both need different time interval triggers to get accurate readings.

VarunKumaran, 2021-08-13:
> the problem seems to be with the writing to the shift register using the setAll() function ... if the example from this driver worked ... than one of our examples should have changed the state of [...]

> @VarunKumaran share me your discord Id . may be i can try to help you their

varunbolt30#9368

VarunKumaran, 2021-08-13:
> the problem seems to be with the writing to the shift register using the setAll() function ... if the example from this driver worked ... than one of our examples should have changed the state of [...]

Okayy give me few mins

itsboo07, 2021-08-13:
> the problem seems to be with the writing to the shift register using the setAll() function ... if the example from this driver worked ... than one of our examples should have changed the state of [...]

yes i feel the same. if the example code worked, one of the above code should have worked. also @VarunKumaran i have sent you request and @sfranzyshen is really super skilled. he might sure help you fix it.

VarunKumaran, 2021-08-13:
#include <ShiftRegister74HC595.h>
ShiftRegister74HC595<1> sr(3, 1, 10);

void setup() {
  sr.setAllLow();
  // put your setup code here, to run once:

}

void loop() {
  
  sr.setAllLow(); // set all pins LOW
  delay(1000); 

  sr.setAllHigh(); // set all pins HIGH
  delay(1000);

  sr.setAllLow(); // set all pins LOW
  delay(1000);

  uint8_t pinValues[] = {B10101010};
  sr.setAll(pinValues);
  delay(2000);

  
  // put your main code here, to run repeatedly:

}

sfranzyshen, 2021-08-13:
Here's two copies of test code that does the same thing two different ways ... I want to test calling the setAll() function multiple times in a row to see if it works ... can you run each of these and post the results?
### 1st
```
#include <ShiftRegister74HC595.h>
ShiftRegister74HC595<1> sr(3, 1, 10);

void setup() {
	sr.setAllLow();
}

void loop() {
	uint8_t pinValues[1];

	sr.setAllHigh(); // set all pins HIGH
	delay(1000);

	sr.setAllLow(); // set all pins LOW
	delay(1000);

	pinValues[0] = B11111111;
	sr.setAll(pinValues);
	delay(1000);

	pinValues[0] = B10101010;
	sr.setAll(pinValues);
	delay(1000);

	pinValues[0] = B00000000;
	sr.setAll(pinValues);
	delay(1000);

}
```
### 2nd
```
#include <ShiftRegister74HC595.h>
ShiftRegister74HC595<1> sr(3, 1, 10);

uint8_t pinValues1[] = {B11111111};
uint8_t pinValues2[] = {B10101010};
uint8_t pinValues3[] = {B00000000};

void setup() {
	sr.setAllLow();
}

void loop() {

	sr.setAllHigh(); // set all pins HIGH
	delay(1000);

	sr.setAllLow(); // set all pins LOW
	delay(1000);

	sr.setAll(pinValues1);
	delay(1000);

	sr.setAll(pinValues2);
	delay(1000);

	sr.setAll(pinValues3);
	delay(1000);
}

```

VarunKumaran, 2021-08-13:
> Here are two copies of test code that does the same thing two different ways ... I want to test calling the setAll() function multiple times in a row to see if it works ... can you run each of these and post the results?
> 
> ### 1st
> ```
> #include <ShiftRegister74HC595.h>
> ShiftRegister74HC595<1> sr(3, 1, 10);
> 
> void setup() {
> 	sr.setAllLow();
> }
> 
> void loop() {
> 	uint8_t pinValues[1];
> 
> 	sr.setAllHigh(); // set all pins HIGH
> 	delay(1000);
> 
> 	sr.setAllLow(); // set all pins LOW
> 	delay(1000);
> 
> 	pinValues[0] = B11111111;
> 	sr.setAll(pinValues);
> 	delay(1000);
> 
> 	pinValues[0] = B10101010;
> 	sr.setAll(pinValues);
> 	delay(1000);
> 
> 	pinValues[0] = B00000000;
> 	sr.setAll(pinValues);
> 	delay(1000);
> 
> }
> ```
> 
> ### 2nd
> ```
> #include <ShiftRegister74HC595.h>
> ShiftRegister74HC595<1> sr(3, 1, 10);
> 
> uint8_t pinValues1[] = {B11111111};
> uint8_t pinValues2[] = {B10101010};
> uint8_t pinValues3[] = {B00000000};
> 
> void setup() {
> 	sr.setAllLow();
> }
> 
> void loop() {
> 
> 	sr.setAllHigh(); // set all pins HIGH
> 	delay(1000);
> 
> 	sr.setAllLow(); // set all pins LOW
> 	delay(1000);
> 
> 	sr.setAll(pinValues1);
> 	delay(1000);
> 
> 	sr.setAll(pinValues2);
> 	delay(1000);
> 
> 	sr.setAll(pinValues3);
> 	delay(1000);
> }
> ```

Sure just gimme a min !!

VarunKumaran, 2021-08-13:
https://user-images.githubusercontent.com/86418940/129420127-faef2bc7-3218-45d0-9b3c-757d75635123.mp4

Test Code 1

VarunKumaran, 2021-08-13:
https://user-images.githubusercontent.com/86418940/129420299-33d7d465-414b-4886-a6e6-343999c41ec7.mp4

Test Code 2
BOTH the code works perfectly fine

itsboo07, 2021-08-13:
if the test code works, @sfranzyshen i feel its somthing to do with the ultrasonic gnd and shift register gnd

VarunKumaran, 2021-08-13:
I will Share my rough Circuit dig

VarunKumaran, 2021-08-13:
[image]

I have not connect any pins to my motion sensor yet so ignore that sketch. As you can see the JSN is my ultrasonic sensor and I have connected them seperate GNDs

itsboo07, 2021-08-13:
@VarunKumaran  try only swaping the LED gnd to pin 24 (GND as per your esp schm) and try the above code with sonic sensors.

VarunKumaran, 2021-08-13:
> @VarunKumaran try only swaping the LED gnd to pin 24 (GND as per your esp schm) and try the above code with sonic sensors.

Tried and same result

sfranzyshen, 2021-08-13:
maybe it has something to do with the pulseIn() ... I have taken the last test code and only added in the reading of the sonic sensor ... lets see what happens with this code ...

```
#include <ShiftRegister74HC595.h>

#define echoPin 4  // attach pin D2 of Esp8266 NodeMcu to pin Echo of HC-SR04
#define trigPin 5  // attach pin D1 of Esp8266 NodeMcu to pin Trig of HC-SR04

// defines variables
long duration;     // variable for the duration of sound wave travel
long d;            // variable for the distance measurement
ShiftRegister74HC595<1> sr(3, 1, 10);
uint8_t pinValues[1];

void setup() {
  pinMode(trigPin, OUTPUT); // Sets the trigPin as an OUTPUT
  pinMode(echoPin, INPUT);  // Sets the echoPin as an INPUT
  Serial.begin(9600);     // Serial Communication is starting with 9600 of baudrate speed
  sr.setAllLow();
}

void loop() {

  sr.setAllHigh(); // set all pins HIGH
  delay(1000);
  get_distance();
  
  sr.setAllLow(); // set all pins LOW
  delay(1000);
  get_distance();

  pinValues[0] = B11111111;
  sr.setAll(pinValues);
  delay(1000);
  get_distance();
  
  pinValues[0] = B10101010;
  sr.setAll(pinValues);
  delay(1000);
  get_distance();

  pinValues[0] = B00000000;
  sr.setAll(pinValues);
  delay(1000);
  get_distance();
  
}

void get_distance() {
  // Clears the trigPin condition
  digitalWrite(trigPin, LOW);
  delayMicroseconds(5);

  // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);

  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);

  // Calculating the distance
  d = (duration * 0.034 / 2) + 3; // Speed of sound wave divided by 2 (go and back)

  // Displays the distance on the Serial Monitor
  Serial.print("Distance: ");
  Serial.print(d);
  Serial.println(" cm");
}

```

VarunKumaran, 2021-08-13:
[image]
Same result bro..

VarunKumaran, 2021-08-13:
[image]
All the leds are light up and doesn't change acc to the distance obtained. 
This was the outcome for all the code we tried.

itsboo07, 2021-08-13:
thats just a test code @VarunKumaran .. @sfranzyshen is checking does the test code works with sonic sensors I guess

VarunKumaran, 2021-08-13:
> thats just a test code @VarunKumaran .. @sfranzyshen is checking does the test code works with sonic sensors I guess

Realised a bit later 😅 and edited my comment.

itsboo07, 2021-08-13:
@VarunKumaran does the leds change as per code? and print distance?

sfranzyshen, 2021-08-13:
> Same result bro..

ok, now remark out the line that calls the pulseIn() function and see if the lights change ...
```
from
  duration = pulseIn(echoPin, HIGH);
to 
 // duration = pulseIn(echoPin, HIGH);

```

VarunKumaran, 2021-08-13:
ok @sfranzyshen

VarunKumaran, 2021-08-13:
https://user-images.githubusercontent.com/86418940/129422769-de60cdb8-e909-49ab-bd2a-57db7dec5c2d.mp4

This was the result after 2 seconds all the led lits up and doesn't go back to low

VarunKumaran, 2021-08-13:
But this time as you can see the first led misbehaves at the start and then all the led lits up and it stays the same

itsboo07, 2021-08-13:
> [image]
> 
> I have not connect any pins to my motion sensor yet so ignore that sketch. As you can see the JSN is my ultrasonic sensor and I have connected them seperate GNDs

@VarunKumaran rather powering the sonic sensor to 3.3 power it to 5v(pin 15 and gnd it to pin 14
and gnd the shift register to pin 10 and try

VarunKumaran, 2021-08-13:
> > [image]
> > I have not connect any pins to my motion sensor yet so ignore that sketch. As you can see the JSN is my ultrasonic sensor and I have connected them seperate GNDs
> 
> @VarunKumaran rather powering the sonic sensor to 3.3v, power it to 5v(pin 15 )and gnd it to pin 14
> and gnd the shift register to pin 10 and try

Same results !! 😔

sfranzyshen, 2021-08-13:
> This was the result after 2 seconds all the led lits up and doesn't go back to low

ok, now remark out **ALL** lines in the get_distance() function that call the digitalWrite() function ... and try again

from:

```
void get_distance() {
  // Clears the trigPin condition
  digitalWrite(trigPin, LOW);
  delayMicroseconds(5);

  // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
```

to:

```
void get_distance() {
  // Clears the trigPin condition
  //digitalWrite(trigPin, LOW);
  delayMicroseconds(5);

  // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
  //digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  //digitalWrite(trigPin, LOW);

```

VarunKumaran, 2021-08-13:
Ok @sfranzyshen

VarunKumaran, 2021-08-13:
> ok, now remark out **ALL** lines in the get_distance() function that call the digitalWrite() function ... and try again
>
> from:
>
> ```
> void get_distance() {
>   // Clears the trigPin condition [...]

Same results as before !!

[39 more replies]

closed 2021-08-14 by VarunKumaran