Simsso/ShiftRegister74HC595 #29: Arduino simulation page for the library 😀

matititam, 2021-09-13:
hi There

Good day

Thank you for the wonderful library for the shift out register.
I am writing this to share with you a link for the Arduino simulation for the useful library you have shared
Permanent link: 

https://wokwi.com/arduino/projects/309636841566372416

[image]

I would love to hear your feedback on the Arduino simulator can be improved so that, people who create and share libraries be benefitted. 
We will be glad, if you also add the link in the ReadMe file so that future readers and visitors will know first-hand the capability of the library. 
Please don't hesitate to ask questions and suggest improvements. 

Thank you!
Matititam

vanhcm20, 2021-11-08:
Can this library use for many 74HC595?

Simsso, 2022-02-01:
That's amazing, thank you @matititam, I'm impressed by the simulator. Will check it out in greater detail :-)

Out of curiosity: does the simulator also support (fake) SD card modules for examples?

matititam, 2022-02-02:
Dear @Simsso It will be awesome fi you can share some feedback. 

How can the simulator be helpful t a developer like you? What else would you love to see in the simulator?

It will be fun, if you can say hi to fellow Wokwi users on [Discord](https://wokwi.com/discord)..at your convenience

closed 2022-02-10 by Simsso